ARMmbed/DAPLink #1005: Make issue on Windows (prebuild and postbuild python scripts issue)

ozersa, 2023-02-03:
On windows side prebuild and post build python script does not able to executed by make. See below.

[image]

The build succeeded, related binaries are generated but python script does not executed.

mbrossard, 2023-02-03:
I started a (CI build for Windows](https://github.com/ARMmbed/DAPLink/actions/workflows/windows.yml) (which does not seem to be using `virtualenv`, see [windows.yml](https://github.com/ARMmbed/DAPLink/blob/main/.github/workflows/windows.yml)) and it seems to work. I am not familiar enough with Python on Windows, but it does seems like the file association for `.py` is not working and the python interpreter is not invoked.

You could also edit `records/tools/gcc_arm.yaml` to replace the `.py` extensions with `.bat`.

The limitations of `pre_build_script` / `post_build_script` in [project_generator](https://github.com/project-generator/project_generator). A related issue is https://github.com/project-generator/project_generator/issues/477 because we can't for instance define `pre_build_script` to be `python %rootdir%/tools/pre_build_script.py`.

ozersa, 2023-02-03:
You build it with "python tools/progen_compile.py..." by github action, and used cmake, it seems work. the issue occurs with step defined in [Section 3.2](https://github.com/ARMmbed/DAPLink/blob/main/docs/DEVELOPERS-GUIDE.md).

The issue seems, make command does not execute python script by shebang directive on windows side, see below screenshot. If I execute it with "shell **python**  XXX.py" the prebuild works, but post build fail in another sub-script...

[image]