ARMmbed/DAPLink #1009: Can nRF52820 HIC firmware be used for nRF52840?

dojyorin, 2023-02-06:
The Nordic nRF528xx series have similar features.
Already released HIC firmware supports 52820 and 52833, but can it be used for 52840 as well?
I have a few 52840 boards and would like to put them to good use.

Thank you🙇

mbrossard, 2023-02-07:
Interesting idea. I think the chances that `nrf52820_bl` and `nrf52820_if` would work unmodified on a nRF52840 board are very small. But it should not be too hard to add the changes to support the nRF52840. What boards did you have in mind?

dojyorin, 2023-02-07:
Thank you for reply.
The boards I was thinking of using are:
- [Adafruit ItsyBitsy nRF52840 Express](https://learn.adafruit.com/adafruit-itsybitsy-nrf52840-express)
- [SwitchScience ISP1807 MCU Board](https://www.switch-science.com/products/6454)

mbrossard, 2023-02-07:
I have the Adafruit Feather (and I like that it has that debug connector), but I was also thinking that the [XIAO BLE](https://wiki.seeedstudio.com/XIAO_BLE) might also be a nice target. Once it works it's pretty easy to change the pin assignment.

dojyorin, 2023-02-08:
Nice, can I request an update so that I can use those HICs in the next release...?

mbrossard, 2023-02-12:
The [nRF52840 Dongle](https://www.nordicsemi.com/Products/Development-hardware/nrf52840-dongle) was also suggested as an interesting target.

With the help from @maxd-nordic, the [feature/nrf52840_hic](https://github.com/mbrossard/DAPLink/tree/feature/nrf52840_hic) branch has been put together. I tested with the Seeed Studio XIAO BLE, but it also has a configuration for the nRF52840 Dongle.

The downside is that all those boards seem to be using very different pinouts, making it impossible to find a common pinout. The default pinout should also work with the smallest package (QFN48 in this case), so a look at the different [nRF52840 package pin assignments](https://infocenter.nordicsemi.com/topic/ps_nrf52840/pin.html?cp=5_0_0_6_0) is warranted.

dojyorin, 2023-02-15:
Thanks for the implementation! I'm happy😊
I now have understanding of the current issues.
I'll try research the pinout of the board I want to use.

dojyorin, 2023-02-16:
**Board: SSCI6454**

- https://www.switch-science.com/products/6454

**High speed digital I/O dedicated pin**

- `P0_08`
- `P0_13`
- `P0_14`
- `P0_15`
- `P0_17`
- `P0_25`
- `P0_26`
- `P0_27`

Onboard general purpose LED: `P0_06`
Onboard general purpose switch: `P1_06`

**Expected code (I don't know if it's right!)**

- `IO_Config.h`

```c
#elif defined(SSCI6454)

#define GPIO_LED_ACTIVE_STATE 0
#define RESET_BUTTON_PULL NRF_GPIO_PIN_PULLUP
#define PIN_SWCLK         NRF_GPIO_PIN_MAP(0, 26) // SWDCLK (Output)
#define PIN_SWDIO         NRF_GPIO_PIN_MAP(0, 14) // SWDIO (Input/Output)
#undef LED_RUNNING                                // Target Running LED (Output)
#undef LED_CONNECTED                              // Connected LED (Output)
#define LED_HID           NRF_GPIO_PIN_MAP(0, 25) // HID LED
#define LED_MSC           NRF_GPIO_PIN_MAP(0, 27) // MSC LED
#define LED_CDC           NRF_GPIO_PIN_MAP(0, 6)  // CDC LED
#undef LED_PWR                                    // POWER LED
#define RESET_BUTTON      NRF_GPIO_PIN_MAP(1, 6)  // Reset button (SW_RESET)
#define UART_TX_PIN       NRF_GPIO_PIN_MAP(0, 15) // UART TX (From IMCU to target)
#define UART_RX_PIN       NRF_GPIO_PIN_MAP(0, 8) // UART RX (From target to IMCU)
#define I2C_SCL_PIN       NRF_GPIO_PIN_MAP(0, 13)  // I2C SCL
#define I2C_SDA_PIN       NRF_GPIO_PIN_MAP(0, 17)  // I2C SDA
```

mbrossard, 2023-02-16:
This is mostly right:
- Unless you want to connect external LEDs, you can just re-use the one you have.
- `P0_18` is the reset pin for the interface, I don't think you can use it as GPIO unless you explicitly disable reset, you should probably assign `PIN_nRESET` to some other pin (it does not have to be high-speed.

If you want to refine your pin assignment:
- Given the maximum UART speed, UART probably does not require high-speed pins.
- The I2C pins are not currently used for anything.

```c
#elif defined(SSCI6454)

#define GPIO_LED_ACTIVE_STATE 0
#define RESET_BUTTON_PULL NRF_GPIO_PIN_PULLUP
#define PIN_SWCLK         NRF_GPIO_PIN_MAP(0, 26) // SWDCLK (Output)
#define PIN_SWDIO         NRF_GPIO_PIN_MAP(0, 14) // SWDIO (Input/Output)
#define PIN_nRESET        NRF_GPIO_PIN_MAP(0, 10) // nRESET Pin
#undef LED_RUNNING                                // Target Running LED (Output)
#undef LED_CONNECTED                              // Connected LED (Output)
#define LED_HID           NRF_GPIO_PIN_MAP(0, 6) // HID LED
#define LED_MSC           NRF_GPIO_PIN_MAP(0, 6) // MSC LED
#define LED_CDC           NRF_GPIO_PIN_MAP(0, 6)  // CDC LED
#undef LED_PWR                                    // POWER LED
#define RESET_BUTTON      NRF_GPIO_PIN_MAP(1, 6)  // SW_RESET
#define UART_TX_PIN       NRF_GPIO_PIN_MAP(0, 15) // UART TX
#define UART_RX_PIN       NRF_GPIO_PIN_MAP(0, 8)  // UART RX
#define I2C_SCL_PIN       NRF_GPIO_PIN_MAP(0, 13) // I2C SCL
#define I2C_SDA_PIN       NRF_GPIO_PIN_MAP(0, 17) // I2C SDA
```

Do not forget to add `SSCI6454` macro definition somewhere. You can add it to `records/hic_hal/nrf52840.yaml` and use the `nrf52840_bl` and `nrf52840_if` projects.

closed 2023-09-20 by dojyorin