ARMmbed/DAPLink #1021: Bulk endpoint compile error for stm32f103xb project for v0257

zhyjason, 2023-03-29:
when I enable BULK_ENDPOINT macro definition with the project of stm32f103xb for v0257, it happened error, like this : 
[image]
is there anyone how to fix it, thanks.

mbrossard, 2023-03-29:
Defining BULK_ENDPOINT (which is what I assume you did) is not enough. The proper way is to include `records/usb/usb-bulk.yaml` in the project or the HIC. You should try something like:

```
diff --git a/projects.yaml b/projects.yaml
index 06c818f4..adae90f1 100644
--- a/projects.yaml
+++ b/projects.yaml
@@ -569,6 +569,7 @@ projects:
     stm32f103xb_stm32f103rb_if:
         - *module_if
         - *module_hic_stm32f103xb
+        - records/usb/usb-bulk.yaml
         - records/board/stm32f103rb.yaml
     stm32f103xb_stm32f207zg_if:
         - *module_if
```

zhyjason, 2023-03-29:
Hi, I just change HID_ENDPOINT to BULK_ENDPOINT in keil MDK, like this : 
[image]
but I compile the porject with HID_ENDPOINT, that is ok.