ARMmbed/DAPLink #1024: ROM USB loader of the LPC11u35

suikan4github, 2023-04-06:
Hi.

There is an unclear description in the lpc11u35.md. Please see the following quote. 

[DAPLink](https://github.com/ARMmbed/DAPLink)/[docs](https://github.com/ARMmbed/DAPLink/tree/main/docs)/[hic](https://github.com/ARMmbed/DAPLink/tree/main/docs/hic)/[lpc11u35.md](https://github.com/ARMmbed/DAPLink/blob/main/docs/hic/lpc11u35.md#memory-map)
```
Because of the flash size limitation, the presence of a ROM USB loader (also Cortex-M0 does not support vector table relocation).
```
This sentence confuses me. Please make it clear.

mathias-arm, 2023-04-06:
You are right something is missing. For the `lpc11u35` the DAPLink Bootloader is not used. Instead we can use the one in ROM. The reduced flash size and lack of vector table relocation would make it a challenge use the DAPLink Bootloader.

suikan4github, 2023-04-09:
Thank you very much for your explanation!

commit 2457a1be (2024-02-13): lpc11u35 doc: Fix comment about bootloader (fixes: #1024)
commit bbb5228d (2024-02-13): lpc11u35 doc: Fix comment about bootloader (fixes: #1024)
closed 2024-02-13 by mathias-arm