ARMmbed/DAPLink #1028: An error is reported after the “progen generate -t uvision” command is executed

15230745073, 2023-06-09:
HHH@RedmiG MINGW64 /d/HHH/DAPLink/DAPLink (main)
$ progen generate -t uvision
progen.generate INFO    Generating uvision for k20dx_bl
progen.generate INFO    Generating uvision for k20dx_ep_agora_if
progen.generate INFO    Generating uvision for k20dx_ep_kairos_if
progen.generate INFO    Generating uvision for k20dx_frdmk20dx_if
progen.generate INFO    Generating uvision for k20dx_frdmk22f_if
progen.generate INFO    Generating uvision for k20dx_frdmk28f_if
progen.generate INFO    Generating uvision for k20dx_frdmk64f_if
progen.generate INFO    Generating uvision for k20dx_frdmk66f_if
progen.generate INFO    Generating uvision for k20dx_frdmk82f_if
progen.generate INFO    Generating uvision for k20dx_frdmkl05z_if
progen.generate INFO    Generating uvision for k20dx_frdmke15z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl25z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl02z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl26z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl03z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl27z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl28z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl43z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl46z_if
progen.generate INFO    Generating uvision for k20dx_frdmkl82z_if
progen.generate INFO    Generating uvision for k20dx_frdmkw24d_if
progen.generate INFO    Generating uvision for k20dx_frdmkw41z_if
progen.generate INFO    Generating uvision for k20dx_hexiwear_if
progen.generate INFO    Generating uvision for k20dx_hvpke18f_if
progen.generate INFO    Generating uvision for k20dx_if
progen.generate INFO    Generating uvision for k20dx_mimxrt1020_evk_qspi_if
progen.generate INFO    Generating uvision for k20dx_mimxrt1050_evk_hyper_if
progen.generate INFO    Generating uvision for k20dx_rapid_iot_if
progen.generate INFO    Generating uvision for k20dx_twrke18f_if
progen.generate INFO    Generating uvision for k20dx_twrkl28z72m_if
progen.generate INFO    Generating uvision for k20dx_xdot_l151_if
progen.generate INFO    Generating uvision for k26f_bl
progen.generate INFO    Generating uvision for k26f_frdmk32w042_if
progen.generate INFO    Generating uvision for k26f_if
progen.generate INFO    Generating uvision for kl26z_artemis_dk_if
progen.generate INFO    Generating uvision for kl26z_bl
progen.generate INFO    Generating uvision for kl26z_if
progen.generate INFO    Generating uvision for kl26z_microbit_if
progen.generate INFO    Generating uvision for kl27z_bl
progen.generate INFO    Generating uvision for kl27z_if
progen.generate INFO    Generating uvision for kl27z_microbit_bl
progen.generate INFO    Generating uvision for kl27z_microbit_if
progen.generate INFO    Generating uvision for lpc11u35_96b_nitrogen_if
progen.generate INFO    Generating uvision for lpc11u35_archble_if
progen.generate INFO    Generating uvision for lpc11u35_archlink_if
progen.generate INFO    Generating uvision for lpc11u35_archmax_if
progen.generate INFO    Generating uvision for lpc11u35_archpro_if
progen.generate INFO    Generating uvision for lpc11u35_blueninja_if
progen.generate INFO    Generating uvision for lpc11u35_c027_if
progen.generate INFO    Generating uvision for lpc11u35_ff1705_l151_if
progen.generate INFO    Generating uvision for lpc11u35_ff_lpc546xx_if
progen.generate INFO    Generating uvision for lpc11u35_gr_lychee_if
progen.generate INFO    Generating uvision for lpc11u35_gr_peach_if
progen.generate INFO    Generating uvision for lpc11u35_hrm1017_if
progen.generate INFO    Generating uvision for lpc11u35_if
progen.generate INFO    Generating uvision for lpc11u35_lpc4088dm_if
progen.generate INFO    Generating uvision for lpc11u35_lpc4088qsb_if
progen.generate INFO    Generating uvision for lpc11u35_lpc812xpresso_if
progen.generate INFO    Generating uvision for lpc11u35_lpc824xpresso_if
progen.generate INFO    Generating uvision for lpc11u35_mini_iot_lpc54018_if
progen.generate INFO    Generating uvision for lpc11u35_musca_a_if
progen.generate INFO    Generating uvision for lpc11u35_musca_b_eflash_if
progen.generate INFO    Generating uvision for lpc11u35_musca_b_if
progen.generate INFO    Generating uvision for lpc11u35_nz32_sc151_if
progen.generate INFO    Generating uvision for lpc11u35_pitaya_link_if
progen.generate INFO    Generating uvision for lpc11u35_rtl8195am_if
progen.generate INFO    Generating uvision for lpc11u35_ssci1114_if
progen.generate INFO    Generating uvision for lpc11u35_ssci824_if
[... 63 lines cut ...]
    files = exporter(self.project['export'], self.settings).export_project()
  File "d:\program files\python\python38\lib\site-packages\project_generator\tools\uvision.py", line 490, in export_project
    path, files = self._export_single_project('uvision') #todo: uvision will switch to uv4
  File "d:\program files\python\python38\lib\site-packages\project_generator\tools\uvision.py", line 452, in _export_single_project
    self._set_target(expanded_dic, uvproj_dic, tool_name)
  File "d:\program files\python\python38\lib\site-packages\project_generator\tools\uvision.py", line 318, in _set_target
    raise RuntimeError("Target %s is not supported. Please add them to https://github.com/project-generator/project_generator_definitions" % expanded_dic['target'].lower())
RuntimeError: Target cortex-m33 is not supported. Please add them to https://github.com/project-generator/project_generator_definitions
"""

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "d:\program files\python\python38\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "d:\program files\python\python38\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "D:\Program Files\Python\Python38\Scripts\progen.exe\__main__.py", line 7, in <module>
  File "d:\program files\python\python38\lib\site-packages\project_generator\__main__.py", line 67, in main
    return args.func(args)
  File "d:\program files\python\python38\lib\site-packages\project_generator\commands\generate.py", line 70, in run
    build_failed, export_failed = r.get(timeout=20.0)
  File "d:\program files\python\python38\lib\multiprocessing\pool.py", line 771, in get
    raise self._value
RuntimeError: Target cortex-m33 is not supported. Please add them to https://github.com/project-generator/project_generator_definitions

carlosperate, 2023-06-15:
Did you install the dependencies from the requirements.txt file?
https://github.com/ARMmbed/DAPLink/blob/main/requirements.txt

What version of `project_generator` are you using?

mathias-arm, 2023-06-15:
The `uvision` projects only support the `armcc` compiler (there is pending work to add support for `armclang`) that does not support Cortex-M33, so it fails trying to while processing LPC55S69 projects.

Try with with specific projects:
```
$ progen generate -t uvision nrf52820_microbit_bl nrf52820_microbit_if
progen.generate INFO	Generating uvision for nrf52820_microbit_bl
progen.generate INFO	Generating uvision for nrf52820_microbit_if
$
```

15230745073, 2023-06-16:
OK thanks, Hope to add support the armclang compiler for uvision projects.

closed 2023-06-20 by mathias-arm