ARMmbed/DAPLink #1033: Not able to build in windows  gcc environment

svsramaraju, 2023-07-18:
Hi , I am not able to build Not able to build in windows  gcc environment . could you please let me know if i am missing anything based on below log

(venv) C:\Projects\DAPLink\GIT_DPLink\DAPLink>python tools\progen_compile.py -t make_gcc_arm  --clean -v --parallel  lpc11u35_if
2023-07-18 15:37:47,463         progen_build INFO       Generating make_gcc_arm for lpc11u35_if
2023-07-18 15:37:47,494       progen.project INFO       Cleaning directory projectfiles\make_gcc_arm\lpc11u35_if
mkdir build
Preprocessing ..\..\..\source\hic_hal\nxp\lpc11u35\gcc\lpc11u35.ld
Compiling arm-none-eabi-cpp -E -x c -P -MMD  -I.  -I..\..\..\source\usb  -I..\..\..\source\daplink\cmsis-dap  -I..\..\..\source\daplink\drag-n-drop  -I..\..\..\source\daplink\usb2uart  -I..\..\..\source\daplink\settings  -I..\..\..\source\daplink  -I..\..\..\source\daplink\interface  -I..\..\..\source\cmsis-core  -I..\..\..\source\hic_hal  -I..\..\..\source\hic_hal\cmsis-driver  -I..\..\..\source\target  -I..\..\..\source\hic_hal\nxp\lpc11u35  -I..\..\..\source\rtos2\Include  -I..\..\..\source\rtos2\RTX\Include -DDAP..LINK_VERSION=259  -DMSC_ENDPOINT  -DCDC_ENDPOINT  -DWEBUSB_INTERFACE  -DWINUSB_INTERFACE  -DDRAG_N_DROP_SUPPORT  -DDAPLINK_IF  -DDAPLINK_BUILD_KEY=0x9B939E8F  -DOS_TICK_FREQ=100  -D__ARM_ARCH_6M__=1  -DINTERFACE_LPC11U35  -DDAPLINK_HIC_ID=0x97969902  -DOS_CLOCK=48000000  -DBULK_ENDPOINT  -D__PROGRAM_START  -DMAIN_TASK_STACK=832  -D__PROGRAM_START  -DMAIN_TASK_STACK=832  ..\..\..\source\hic_hal\nxp\lpc11u35\gcc\lpc11u35.ld -o build/lpc11u35_if.generated.ld
\..\..\source\daplink/HardFault_Handler.c
Compiling ..\..\..\source\daplink/circ_buf.c
Compiling ..\..\..\source\daplink\cmsis-dap/DAP.c
Compiling ..\..\..\source\daplink\cmsis-dap/DAP_UART.c
process_begin: CreateProcess(NULL, arm-none-eabi-cpp -E -x c -P -MMD -I. -I..\..\..\source\usb -I..\..\..\source\daplink\cmsis-dap -I..\..\..\source\daplink\drag-n-drop -I..\..\..\source\daplink\usb2uart -I..\..\..\source\daplink\settings -I..\..\..\source\daplink -I..\..\..\source\daplink\interface -I..\..\..\source\cmsis-core -I..\..\..\source\hic_hal -I..\..\..\source\hic_hal\cmsis-driver -I..\..\..\source\target -I..\..\..\source\hic_hal\nxp\lpc11u35 -I..\..\..\source\rtos2\Include -I..\..\..\source\rtos2\RTX\Include -DDAPLINK_VERSION=259 -DMSC_ENDPOINT -DCDC_ENDPOINT -DWEBUSB_INTERFACE -DWINUSB_INTERFACE -DDRAG_N_DROP_SUPPORT -DDAPLINK_IF -DDAPLINK_BUILD_KEY=0x9B939E8F -DOS_TICK_FREQ=100 -D__ARM_ARCH_6M__=1 -DINTERFACE_LPC11U35 -DDAPLINK_HIC_ID=0x97969902 -DOS_CLOCK=48000000 -DBULK_ENDPOINT -D__PROGRAM_START -DMAIN_TASK_STACK=832 -D__PROGRAM_START -DMAIN_TASK_STACK=832 ..\..\..\source\hic_hal\nxp\lpc11u35\gcc\lpc11u35.ld -o build/lpc11u35_if.generated.ld, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:278: build/lpc11u35_if.generated.ld] Error 2
make: *** Waiting for unfinished jobs....
arm-none-eabi-gcc -Os  -g3  -Werror  -ffunction-sections  -fdata-sections  -std=gnu99  -fshort-wchar  -fno-common  -ffreestanding  -fverbose-asm  -mno-unaligned-access  -Wa,-adln=build/HardFault_Handler.o.lst  -mcpu=cortex-m0 -mthumb  -I.  -I..\..\..\source\usb  -I..\..\..\source\daplink\cmsis-dap  -I..\..\..\source\daplink\drag-n-drop  -I..\..\..\source\daplink\usb2uart  -I..\..\..\source\daplink\settings  -I..\..\..\source\daplink  -I..\..\..\source\daplink\interface  -I..\..\..\source\cmsis-core  -I..\..\..\source\hic_hal  -I..\..\..\source\hic_hal\cmsis-driver  -I..\..\..\source\target  -I..\..\..\source\hic_hal\nxp\lpc11u35  -I..\..\..\source\rtos2\Include  -I..\..\..\source\rtos2\RTX\Include -DDAPLINK_VERSION=259  -DMSC_ENDPOINT  -DCDC_ENDPOINT  -DWEBUSB_INTERFACE  -DWINUSB_INTERFACE  -DDRAG_N_DROP_SUPPORT  -DDAPLINK_IF  -DDAPLINK_BUILD_KEY=0x9B939E8F  -DOS_TICK_FREQ=100  -D__ARM_ARCH_6M__=1  -DINTERFACE_LPC11U35  -DDAPLINK_HIC_ID=0x97969902  -DOS_CLOCK=48000000  -DBULK_ENDPOINT  -D__PROGRAM_START  -DMAIN_TASK_STACK=832  -D__PROGRAM_START  -DMAIN_TASK_STACK=832  -c -MMD -MP ..\..\..\source\daplink/HardFault_Handler.c -o build/HardFault_Handler.o
process_begin: CreateProcess(NULL, arm-none-eabi-gcc -Os -g3 -Werror -ffunction-sections -fdata-sections -std=gnu99 -fshort-wchar -fno-common -ffreestanding -fverbose-asm -mno-unaligned-access -Wa,-adln=build/HardFault_Handler.o.lst -mcpu=cortex-m0 -mthumb -I. -I..\..\..\source\usb -I..\..\..\source\daplink\cmsis-dap -I..\..\..\source\daplink\drag-n-drop -I..\..\..\source\daplink\usb2uart -I..\..\..\source\daplink\settings -I..\..\..\source\daplink -I..\..\..\source\daplink\interface -I..\..\..\source\cmsis-core -I..\..\..\source\hic_hal -I..\..\..\source\hic_hal\cmsis-driver -I..\..\..\source\target -I..\..\..\source\hic_hal\nxp\lpc11u35 -I..\..\..\source\rtos2\Include -I..\..\..\source\rtos2\RTX\Include -DDAPLINK_VERSION=259 -DMSC_ENDPOINT -DCDC_ENDPOINT -DWEBUSB_INTERFACE -DWINUSB_INTERFACE -DDRAG_N_DROP_SUPPORT -DDAPLINK_IF -DDAPLINK_BUILD_KEY=0x9B939E8F -DOS_TICK_FREQ=100 -D__ARM_ARCH_6M__=1 -DINTERFACE_LPC11U35 -DDAPLINK_HIC_ID=0x97969902 -DOS_CLOCK=48000000 -DBULK_ENDPOINT -D__PROGRAM_START -DMAIN_TASK_STACK=832 -D__PROGRAM_START -DMAIN_TASK_STACK=832 -c -MMD -MP ..\..\..\source\daplink/HardFault_Handler.c -o build/HardFault_Handler.o, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:223: build/HardFault_Handler.o] Error 2
arm-none-eabi-gcc -Os  -g3  -Werror  -ffunction-sections  -fdata-sections  -std=gnu99  -fshort-wchar  -fno-common  -ffreestanding  -fverbose-asm  -mno-unaligned-access  -Wa,-adln=build/circ_buf.o.lst  -mcpu=cortex-m0 -mthumb  -I.  -I..\..\..\source\usb  -I..\..\..\source\daplink\cmsis-dap  -I..\..\..\source\daplink\drag-n-drop  -I..\..\..\source\daplink\usb2uart  -I..\..\..\source\daplink\settings  -I..\..\..\source\daplink  -I..\..\..\source\daplink\interface  -I..\..\..\source\cmsis-core  -I..\..\..\source\hic_hal  -I..\..\..\source\hic_hal\cmsis-driver  -I..\..\..\source\target  -I..\..\..\source\hic_hal\nxp\lpc11u35  -I..\..\..\source\rtos2\Include  -I..\..\..\source\rtos2\RTX\Include -DDAPLINK_VERSION=259  -DMSC_ENDPOINT  -DCDC_ENDPOINT  -DWEBUSB_INTERFACE  -DWINUSB_INTERFACE  -DDRAG_N_DROP_SUPPORT  -DDAPLINK_IF  -DDAPLINK_BUILD_KEY=0x9B939E8F  -DOS_TICK_FREQ=100  -D__ARM_ARCH_6M__=1  -DINTERFACE_LPC11U35  -DDAPLINK_HIC_ID=0x97969902  -DOS_CLOCK=48000000  -DBULK_ENDPOINT  -D__PROGRAM_START  -DMAIN_TASK_STACK=832  -D__PROGRAM_START  -DMAIN_TASK_STACK=832  -c -MMD -MP ..\..\..\source\daplink/circ_buf.c -o build/circ_buf.o
process_begin: CreateProcess(NULL, arm-none-eabi-gcc -Os -g3 -Werror -ffunction-sections -fdata-sections -std=gnu99 -fshort-wchar -fno-common -ffreestanding -fverbose-asm -mno-unaligned-access -Wa,-adln=build/circ_buf.o.lst -mcpu=cortex-m0 -mthumb -I. -I..\..\..\source\usb -I..\..\..\source\daplink\cmsis-dap -I..\..\..\source\daplink\drag-n-drop -I..\..\..\source\daplink\usb2uart -I..\..\..\source\daplink\settings -I..\..\..\source\daplink -I..\..\..\source\daplink\interface -I..\..\..\source\cmsis-core -I..\..\..\source\hic_hal -I..\..\..\source\hic_hal\cmsis-driver -I..\..\..\source\target -I..\..\..\source\hic_hal\nxp\lpc11u35 -I..\..\..\source\rtos2\Include -I..\..\..\source\rtos2\RTX\Include -DDAPLINK_VERSION=259 -DMSC_ENDPOINT -DCDC_ENDPOINT -DWEBUSB_INTERFACE -DWINUSB_INTERFACE -DDRAG_N_DROP_SUPPORT -DDAPLINK_IF -DDAPLINK_BUILD_KEY=0x9B939E8F -DOS_TICK_FREQ=100 -D__ARM_ARCH_6M__=1 -DINTERFACE_LPC11U35 -DDAPLINK_HIC_ID=0x97969902 -DOS_CLOCK=48000000 -DBULK_ENDPOINT -D__PROGRAM_START -DMAIN_TASK_STACK=832 -D__PROGRAM_START -DMAIN_TASK_STACK=832 -c -MMD -MP ..\..\..\source\daplink/circ_buf.c -o build/circ_buf.o, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:223: build/circ_buf.o] Error 2
arm-none-eabi-gcc -Os  -g3  -Werror  -ffunction-sections  -fdata-sections  -std=gnu99  -fshort-wchar  -fno-common  -ffreestanding  -fverbose-asm  -mno-unaligned-access  -Wa,-adln=build/DAP.o.lst  -mcpu=cortex-m0 -mthumb  -I.  -I..\..\..\source\usb  -I..\..\..\source\daplink\cmsis-dap  -I..\..\..\source\daplink\drag-n-drop  -I..\..\..\source\daplink\usb2uart  -I..\..\..\source\daplink\settings  -I..\..\..\source\daplink  -I..\..\..\source\daplink\interface  -I..\..\..\source\cmsis-core  -I..\..\..\source\hic_hal  -I..\..\..\source\hic_hal\cmsis-driver  -I..\..\..\source\target  -I..\..\..\source\hic_hal\nxp\lpc11u35  -I..\..\..\source\rtos2\Include  -I..\..\..\source\rtos2\RTX\Include -DDAPLINK_VERSION=259  -DMSC_ENDPOINT  -DCDC_ENDPOINT  -DWE [... cut ...] _TASK_STACK=832 -c -MMD -MP ..\..\..\source\daplink\cmsis-dap/DAP_UART.c -o build/DAP_UART.o, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [Makefile:223: build/DAP_UART.o] Error 2
2023-07-18 15:37:51,866  progen.tools.gccarm ERROR      Project: errors build failed with the status: projectfiles\make_gcc_arm\lpc11u35_if\Makefile
2023-07-18 15:37:51,866         progen_build ERROR      Error building project lpc11u35_if

(venv) C:\Projects\DAPLink\GIT_DPLink\DAPLink>

kkitayam, 2023-07-19:
Does your PATH environment variable include to arm-none-eabi-gcc? If the PATH does not include arm-none-eabi-gcc, `CreateProcess` fails.
You can check whether included or not by using `where arm-none-eabi-gcc` command on command prompt.

svsramaraju, 2023-07-19:
it's PATH issue. I have corrected and able to build. thank you very much for the response.

svsramaraju, 2023-07-19:
Could you please close the issue?

mathias-arm, 2023-07-19:
Thank you @kkitayam for the help.

closed 2023-07-19 by mathias-arm