ARMmbed/DAPLink #1034: GNU Arm Embedded Toolchain is deprecated

terrancewong, 2023-07-22:
`DAPLink/tree/main/docs/DEVELOPERS-GUIDE.md` says 
```
Install a compiler:
    [GNU Arm Embedded Toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) . This compiler will be identified as gcc_arm
```

But this page says 

```
This page is deprecated
See Arm GNU Toolchain for updated content and latest releases

[Arm GNU Toolchain](https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain)
```

BUT all these build scripts does not work with Arm GNU Toolchain. So we must use a deprecated toolchain?

kkitayam, 2023-07-22:
Arm GNU toolchaint is he successor to the GNU Arm Embedded Toolchain.
We should simply use Arm GNU toolchain. Please find new download page at https://developer.arm.com/downloads/-/arm-gnu-toolchain-downloads
We should update the link.

mathias-arm, 2023-07-25:
There are a couple of issues that prevent DAPLink from using Arm GNU toolchain (except the 11.2 release if I remember correctly). The details are listed in #1003. The main issue is that we have not found a way to support newer and older versions of GCC at the same time.

The recommendation for the time being is to continue using the 10.3 version which still works very well for our needs. Eventually, after a release and extensive testing we will switch to require a newer version of GCC.

terrancewong, 2023-07-28:
I build successfully built it with latest  arm-none-eabi-gcc 12.2

mathias-arm, 2023-08-03:
Closing in favor of https://github.com/ARMmbed/DAPLink/issues/1040

closed 2023-08-03 by mathias-arm