ARMmbed/DAPLink #1037: Building on Mac with latest homebrew-managed `gcc-arm-embedded` fails

fughilli, 2023-07-25:
The latest `gcc-arm-embedded` from Homebrew provides GCC 12.2.1. This version complains when trying to build the SAM3U2C bootloader and interface firmwares:

```
> python3 tools/progen_compile.py -t make_gcc_arm --parallel sam3u2c_if sam3u2c_bl --clean
Preprocessing ../../../source/daplink/daplink.ld
Compiling ../../../source/daplink/HardFault_Handler.c
Compiling ../../../source/daplink/circ_buf.c
Compiling ../../../source/daplink/cmsis-dap/DAP.c
Compiling ../../../source/daplink/cmsis-dap/DAP_UART.c
Compiling ../../../source/daplink/cmsis-dap/DAP_queue.c
Compiling ../../../source/daplink/cmsis-dap/JTAG_DP.c
Compiling ../../../source/daplink/cmsis-dap/DAP_vendor.c
Compiling ../../../source/daplink/cmsis-dap/SWO.c
Compiling ../../../source/daplink/cmsis-dap/SW_DP.c
Compiling ../../../source/daplink/cmsis-dap/daplink_swd_dynamic_pins.c
Compiling ../../../source/daplink/cortex_m.c
Compiling ../../../source/daplink/crc32.c
Compiling ../../../source/daplink/daplink_debug.c
Compiling ../../../source/daplink/drag-n-drop/file_stream.c
Compiling ../../../source/daplink/drag-n-drop/flash_decoder.c
Compiling ../../../source/daplink/drag-n-drop/flash_intf.c
Compiling ../../../source/daplink/drag-n-drop/flash_manager.c
Compiling ../../../source/daplink/drag-n-drop/iap_flash_intf.c
Compiling ../../../source/daplink/drag-n-drop/intelhex.c
Compiling ../../../source/daplink/drag-n-drop/vfs_user.c
Compiling ../../../source/daplink/drag-n-drop/vfs_manager.c
Compiling ../../../source/daplink/drag-n-drop/virtual_fs.c
Compiling ../../../source/daplink/error.c
Compiling ../../../source/daplink/flash_hal.c
Compiling ../../../source/daplink/info.c
Compiling ../../../source/daplink/interface/bootloader_update.c
Compiling ../../../source/daplink/interface/daplink.c
Compiling ../../../source/daplink/interface/main_interface.c
Compiling ../../../source/daplink/interface/swd_host_ca.c
Compiling ../../../source/daplink/interface/swd_host.c
Compiling ../../../source/daplink/interface/target_flash.c
Compiling ../../../source/daplink/sdk_stub.c
Compiling ../../../source/daplink/settings/settings.c
Compiling ../../../source/daplink/settings/settings_rom.c
Compiling ../../../source/daplink/util.c
Compiling ../../../source/daplink/usb2uart/usbd_user_cdc_acm.c
Compiling ../../../source/family/freescale/target_reset_Kseries.c
Compiling ../../../source/daplink/validation.c
Compiling ../../../source/family/freescale/target_reset_Lseries.c
Compiling ../../../source/family/freescale/target_reset_mimxrt.c
Compiling ../../../source/family/freescale/target_reset_rapid_iot.c
Compiling ../../../source/family/nordic/target_reset_nrf51.c
Compiling ../../../source/family/nordic/target_reset_nrf52.c
Compiling ../../../source/family/realtek/target_reset_realtek_rtl8195am.c
Compiling ../../../source/family/renesas/target_reset_rza.c
Compiling ../../../source/family/toshiba/target_reset_tz.c
Compiling ../../../source/hic_hal/atmel/sam3u2c/flash_hal_SAM3U.c
Compiling ../../../source/hic_hal/atmel/sam3u2c/gpio.c
Compiling ../../../source/hic_hal/atmel/sam3u2c/read_uid.c
Compiling ../../../source/hic_hal/atmel/sam3u2c/system_SAM3U.c
Compiling ../../../source/hic_hal/atmel/sam3u2c/uart.c
Compiling ../../../source/hic_hal/atmel/sam3u2c/usb_config.c
Compiling ../../../source/hic_hal/atmel/sam3u2c/usbd_ATSAM3U2C.c
Compiling ../../../source/rtos2/RTX/Config/RTX_Config.c
Compiling ../../../source/rtos2/RTX/Source/rtx_delay.c
Compiling ../../../source/rtos2/RTX/Source/rtx_evr.c
Compiling ../../../source/rtos2/RTX/Source/rtx_kernel.c
Compiling ../../../source/rtos2/RTX/Source/rtx_lib.c
Compiling ../../../source/rtos2/RTX/Source/rtx_mempool.c
Compiling ../../../source/rtos2/RTX/Source/rtx_memory.c
Compiling ../../../source/rtos2/RTX/Source/rtx_mutex.c
Compiling ../../../source/rtos2/RTX/Source/rtx_msgqueue.c
Compiling ../../../source/rtos2/RTX/Source/rtx_system.c
Compiling ../../../source/rtos2/RTX/Source/rtx_thread.c
Compiling ../../../source/rtos2/RTX/Source/rtx_timer.c
Compiling ../../../source/rtos2/Source/os_systick.c
Compiling ../../../source/target/target_board.c
Compiling ../../../source/target/target_default.c
Compiling ../../../source/target/target_family.c
Compiling ../../../source/usb/cdc/usbd_core_cdc.c
Compiling ../../../source/usb/cdc/usbd_cdc_acm.c
Compiling ../../../source/usb/msc/usbd_core_msc.c
Compiling ../../../source/usb/msc/usbd_msc.c
Compiling ../../../source/usb/usbd_core.c
Compiling ../../../source/usb/webusb/usbd_core_webusb.c
Compiling ../../../source/usb/winusb/usbd_core_winusb.c
Assembling ../../../source/rtos2/RTX/Source/GCC/irq_cm3.S
Assembling ../../../source/hic_hal/atmel/sam3u2c/gcc/startup_SAM3U.S
Linking build/sam3u2c_if.elf
/Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-closer.o): in function `_close_r':
closer.c:(.text._close_r+0xc): warning: _close is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-lseekr.o): in function `_lseek_r':
lseekr.c:(.text._lseek_r+0x10): warning: _lseek is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-readr.o): in function `_read_r':
readr.c:(.text._read_r+0x10): warning: _read is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: /Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/lib/thumb/v7-m/nofp/libc_nano.a(libc_a-writer.o): in function `_write_r':
writer.c:(.text._write_r+0x10): warning: _write is not implemented and will always fail
/Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: warning: build/sam3u2c_if.elf has a LOAD segment with RWX permissions
/Applications/ArmGNUToolchain/12.2.mpacbti-rel1/arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld: warning: build/sam3u2c_if.elf has a LOAD segment with RWX permissions
Memory region         Used Size  Region Size  %age Used
    m_interrupts:          1 KB         1 KB    100.00%
          m_text:         91 KB        91 KB    100.00%
        m_cfgrom:          10 B         4 KB      0.24%
          m_data:       23104 B      32512 B     71.06%
        m_cfgram:         142 B        256 B     55.47%
collect2: error: ld returned 1 exit status
make: *** [build/sam3u2c_if.elf] Error 1
2023-07-24 14:59:12,112  progen.tools.gccarm ERROR      Project: errors build failed with the status: projectfiles/make_gcc_arm/sam3u2c_if/Makefile
2023-07-24 14:59:12,112         progen_build ERROR      Error building project sam3u2c_if
```

fughilli, 2023-07-25:
Just realized this is a dup of #1003

mathias-arm, 2023-08-03:
Closing in favor of #1040

closed 2023-08-03 by mathias-arm