ARMmbed/DAPLink #1071: [miro:bit] [0258] PWR led turns sometimes off when device is powered by battery on reset button press

fabianhugo, 2024-04-18:
I noticed that on the v0258 beta3 firmware from micro:bit v2.21the power led sometimes turns off when I press the reset button. So the device looks off, while it is actually still powered. This bug also concerns the Calliope mini V3 which uses DAPLink version 0258, and sometimes confuses users.

mathias-arm, 2024-04-18:
Pinging @microbit-carlos

microbit-carlos, 2024-04-19:
Hi @fabianhugo
Could you provide exact steps to reproduce?

If the reset button is pressed around 5 seconds (you can see the orange LED dimming out), the board is meant to go into a "deep sleep" mode, and in that mode the power red LED will turn off. Is that what you meant? Or does it turn off in a different situation?

fabianhugo, 2024-04-19:
Hi @microbit-carlos, 
thanks for your fast reply :) 
Yes, i always also wondering if it has something to do with the deep sleep mode, but the main program keeps running and I am pressing the reset button just very briefly. In programs where the display is used this is not a real problem, but it can be, when the display remains off. 
I checked the interface firmware in DAPLink version v0257 and the problem is not there. Below I attached two videos with a micro:bit v2.20. The program on the micro:bit is a standard makecode program:https://makecode.microbit.org/_95gfJ5TskDvb 

### DAPLink v0258 IF

https://github.com/ARMmbed/DAPLink/assets/32654742/981174ee-5380-49e5-b721-e8f6d6a8c0c7

### DAPLink v0257 IF

https://github.com/ARMmbed/DAPLink/assets/32654742/d429392b-a06e-403c-a3ba-4a823a743f29