ARMmbed/DAPLink #1073: Clarification on flash blobs

vincent290587, 2024-04-21:
Hi all, 

I am currently adding support for STM32F723 (USB HS) based Daplink.

My target is a STM32H7 chip with external memories (over SPI), I would like to be able to program both the internal memory and the external memories of the H7 target.

- Is that even possible ?
- Is there documentation on how the programming (e.g. in OpenOCD differs in order to program internal/external ?

If so I assume the FLashBlob project is the way to go, correct ?

mbrossard, 2024-04-21:
I know of one example that has two flash algos, one of which is QSPI:
- This where the multiple flash algos are defined: https://github.com/ARMmbed/DAPLink/blob/main/source/family/arm/musca_b/flash_blob.c 
- You can define the target with multiple regions and specify for each which flash algo to use: https://github.com/ARMmbed/DAPLink/blob/main/source/family/arm/musca_b/target.c

Source for QSPI flash algo: https://github.com/pyocd/FlashAlgo/tree/main/source/arm/mt25ql512

mathias-arm, 2025-02-27:
Were you successful in adding support for STM32F723?

closed 2025-02-27 by mathias-arm

vincent290587, 2025-03-14:
Mostly, but USB HS is not fully tested yet