ARMmbed/DAPLink #1083: macOS 15 Sequoia finder drag&drop flashing fails

microbit-carlos, 2024-10-17:
Not quite the same, but similar to:
- https://github.com/ARMmbed/DAPLink/issues/982

Some things we've noticed so far:
- There is a lot of USB activity when the drive is mounted, 20 to 60 seconds (might depend on computer speed, a colleague with a mac mini sees this for 20 seconds, the slow VM I use for testing takes 60 seconds).
- If a hex file is flashed during the activity period, the flash success
- If a hex file is flashed after this activity is done, the flash fails

fabianhugo, 2024-10-23:
I am seeing something very similar on Ubuntu 22.04: uname -a shows: 
`Linux hugoUbu20 6.8.0-45-generic #45~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Wed Sep 11 15:25:05 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`

Copying of universal hex often fails with error 504 and the fail.txt reads:
```
error: The transfer timed out.
type: transient, user
```
Around 3 of 4 file transfer fail. Using cp or rsync in the terminal do also not improve the behavior. And WebUSB is not affected as well.

One way for me to tell that the copying failed is to check the contents of the mass storage. If the hex is visible there, it failed.
[image]

martinwork, 2024-10-23:
@fabianhugo 
For Linux, this UDEV rules file [99-microbit-udisks.zip](https://github.com/user-attachments/files/17497418/99-microbit-udisks.zip) can help by adding the "sync" option to MICROBIT drives.

To install...

Connect micro:bit and check current options:
`mount | grep MICROBIT`

Unzip 99-microbit-udisks.zip, and copy the rules file to the udev rules.d folder. For example
`sudo cp 99-microbit-udisks.rules /etc/udev/rules.d`

Check it has the same owner, group and permissions as other rules files.
`ls -l /etc/udev/rules.d`

Restart UDEV.
`sudo udevadm control --reload-rules ; sudo udevadm trigger`

Disconnect and reconnect micro:bit then check what options are in use now.
`mount | grep MICROBIT`

The only difference should be the addition of "sync".

To uninstall, delete 99-microbit-udisks.rules from the rules.d folder.

fabianhugo, 2024-10-28:
Seems to have fixed it! Thank you! :)

microbit-carlos, 2024-11-08:
## Analysis

With macOS 14 and older, hex files were written by the OS into continuous FAT sectors. On macOS 15 the hex file blocks are sent in order, but they are not written into continuous filesystem sectors. This is due to fragmentation introduced by previously created metadata files, the reason the orange LED blinks for so long after the USB drive is mounted.

I added additional RTT logging into a DAPLink test build, to capture the filesystem traffic data to check what data blocks are written into what the file system sectors. The following files have been captured:

- [drive-mount.log](https://github.com/user-attachments/files/17682575/drive-mount.log): captures the filesystem traffic when the micro:bit USB is connected to the computer, macOS mounts the drive, and starts creating metadata files.
- [flashing-ih-9903.log](https://github.com/user-attachments/files/17682576/flashing-ih-9903.log): captures the filesystem traffic when the `ih-9903.txt` file was copied into the micro:bit USB drive
- [ih-9903.hex.zip](https://github.com/user-attachments/files/17682579/ih-9903.hex.zip): is the hex file copied into the USB drive

In these files we can see the following:

- The `flashing-ih-9903.txt` shows the Hex file data is being sent in order
- The first file block is written into filesystem sector 373
- It writes 32 file blocks up to filesystem sector 404, and then the following block is sent to sector 413
- Then it continues writing the file in continuous sector, until sector 452, where it jumps to 549
- It does a few more jumps like that for the rest of the file
- Looking at `drive-mount.log` we can see that the OS had previously already written some kind metadata file content into sector 407 (first jump), in sectors 453 to 455, sectors 461 to 475 (second jump), etc.

This suggests that the hex file blocks are written into non-consecutive filesystem sectors because the OS has already created several metadata files on the drive occupying the sectors that would have otherwise be used by the hex file.

This is the file structure of the micro:bit USB drive in macOS 12, where no metadata files are created by the OS:
```
📂 /Volumes/MICROBIT
┣━━ 📄 DETAILS.TXT (592 bytes)
┣━━ 📄 FAIL.TXT (55 bytes)
┗━━ 📄 MICROBIT.HTM (284 bytes)
```

In contrast, this is the file structure of the same micro:bit USB drive in macOS 15:
```

📂 /Volumes/NO NAME
┣━━ 📂 .fseventsd
┃   ┗━━ 📄 fseventsd-uuid (36 bytes)
┣━━ 📂 .Spotlight-V100
┃   ┣━━ 📂 Store-V2
┃   ┃   ┗━━ 📂 B8624886-8BF7-4584-8688-AEBC5B953367
┃   ┃       ┣━━ 📂 journals.assisted_import_post
┃   ┃       ┣━━ 📂 journals.assisted_import_pre
┃   ┃       ┣━━ 📂 journals.corespotlight
┃   ┃       ┣━━ 📂 journals.health_check
┃   ┃       ┣━━ 📂 journals.live
┃   ┃       ┣━━ 📂 journals.live_priority
┃   ┃       ┣━━ 📂 journals.live_system
┃   ┃       ┣━━ 📂 journals.live_user
┃   ┃       ┣━━ 📂 journals.migration
┃   ┃       ┣━━ 📂 journals.migration_secondchance
┃   ┃       ┣━━ 📂 journals.scan
┃   ┃       ┣━━ 📄 .store.db (36.9 kB)
┃   ┃       ┣━━ 📄 0.directoryStoreFile (65.5 kB)
┃   ┃       ┣━━ 📄 0.directoryStoreFile.shadow (1.2 kB)
┃   ┃       ┣━━ 📄 0.indexArrays (73.0 kB)
┃   ┃       ┣━━ 📄 0.indexBigDates (20 bytes)
┃   ┃       ┣━━ 📄 0.indexCompactDirectory (1.1 kB)
┃   ┃       ┣━━ 📄 0.indexDirectory (2.1 kB)
┃   ┃       ┣━━ 📄 0.indexGroups (3.3 kB)
┃   ┃       ┣━━ 📄 0.indexHead (4.1 kB)
┃   ┃       ┣━━ 📄 0.indexId (1 byte)
┃   ┃       ┣━━ 📄 0.indexIds (40 bytes)
┃   ┃       ┣━━ 📄 0.indexPositions (281 bytes)
┃   ┃       ┣━━ 📄 0.indexPostings (1.1 kB)
┃   ┃       ┣━━ 📄 0.shadowIndexGroups (4 bytes)
┃   ┃       ┣━━ 📄 0.shadowIndexHead (4.1 kB)
┃   ┃       ┣━━ 📄 Cab.created (0 bytes)
┃   ┃       ┣━━ 📄 dbStr-1.map.buckets (4.1 kB)
┃   ┃       ┣━━ 📄 dbStr-1.map.data (4.1 kB)
┃   ┃       ┣━━ 📄 dbStr-1.map.header (56 bytes)
┃   ┃       ┣━━ 📄 dbStr-1.map.header.shadow (56 bytes)
┃   ┃       ┣━━ 📄 dbStr-1.map.offsets (4.1 kB)
[... 44 lines cut ...]
┃   ┃       ┣━━ 📄 live.0.shadowIndexCompactDirectory (8 bytes)
┃   ┃       ┣━━ 📄 live.0.shadowIndexDirectory (2.1 kB)
┃   ┃       ┣━━ 📄 live.0.shadowIndexGroups (1 byte)
┃   ┃       ┣━━ 📄 live.0.shadowIndexHead (4.1 kB)
┃   ┃       ┣━━ 📄 live.0.shadowIndexPositionTable (0 bytes)
┃   ┃       ┣━━ 📄 live.0.shadowIndexTermIds (0 bytes)
┃   ┃       ┣━━ 📄 psid.db (8.2 kB)
┃   ┃       ┣━━ 📄 reverseDirectoryStore (65.5 kB)
┃   ┃       ┣━━ 📄 reverseDirectoryStore.shadow (3.1 kB)
┃   ┃       ┣━━ 📄 reverseStore.updates (2 bytes)
┃   ┃       ┣━━ 📄 shutdown_time (4 bytes)
┃   ┃       ┣━━ 📄 store.db (36.9 kB)
┃   ┃       ┣━━ 📄 store.updates (3 bytes)
┃   ┃       ┣━━ 📄 store_generation (4 bytes)
┃   ┃       ┣━━ 📄 tmp.Cab (0 bytes)
┃   ┃       ┣━━ 📄 tmp.Glow (0 bytes)
┃   ┃       ┣━━ 📄 tmp.Lion (0 bytes)
┃   ┃       ┣━━ 📄 tmp.spotlight.loc (9.1 kB)
┃   ┃       ┣━━ 📄 tmp.spotlight.state (4.1 kB)
┃   ┃       ┗━━ 📄 tmp.Star (0 bytes)
┃   ┗━━ 📄 VolumeConfiguration.plist (4.5 kB)
┣━━ 📄 DETAILS.TXT (592 bytes)
┣━━ 📄 FAIL.TXT (55 bytes)
┗━━ 📄 MICROBIT.HTM (284 bytes)
```