ARMmbed/DAPLink #1085: DAPlink with LPC11U35 (Fastening the writing process).

fabio-ric-silva, 2024-11-21:
Someone here has this kind of interface with LPC11U35 and was able for fastening the programming process?

I have been studying how to implement such modification, but having no success!

Could, someone helping me with it!

The idea is using just as writing interfacing for pyocd project!

Sincerely;

mathias-arm, 2024-12-05:
If you want to speed up programming with `lpc11u35`, you should use or create a DAPLink project that support drag-n-drop flashing for your target. This is what give the best performance.

There are two options HIC options for DAP transfers with `lpc11u35`: `module_hic_lpc11u35` which uses HID and `module_hic_lpc11u35_bulk`  which uses bulk. It's not possible to have both at the same time as CDC and MSC because of the limited number of endpoints. The `lpc11u35_if` uses bulk (`module_hic_lpc11u35_bulk`) for DAP and in my tests works with PyOCD.

If you want to support CMSIS-DAPv2 (bulk) with `lpc11u35` with another project (e.g. `lpc11u35_pitaya_link_if`) you should be able to replace `module_hic_lpc11u35` with `module_hic_lpc11u35_bulk` for the project.

fabio-ric-silva, 2024-12-06:
> If you want to speed up programming with `lpc11u35`, you should use or create a DAPLink project that support drag-n-drop flashing for your target. This is what give the best performance.
>
> There are [...]

Mathias, actually, I just need the CMSIS-DAP running on bulk, and being recognized by the PYOcd! There is no need of the others interfaces! I have made a test, take off all the interfaces, keeping just the CMSIS-DAP, ant taking off the HID, CDC, WEB. Even though the PYOcd couldn't recognize the CMSIS-DAP in bulk mode!

mathias-arm, 2024-12-06:
> The `lpc11u35_if` uses bulk (`module_hic_lpc11u35_bulk`) for DAP and in my tests works with PyOCD.
Test the `lpc11u35_if` project without modifying it. It does use bulk mode, but it is not guaranteed to be that much faster than HID if PyOCD cannot leverage the larger packet size.