ARMmbed/DAPLink #1086: progen generate -f projects.yaml -p lpc55s69_mculink_if -t uvision, problem!

fabio-ric-silva, 2024-12-03:
Dears, I just gave up from speeding up the LPC11U HIC.

 Studying a better time more, I found a possibility of using Usb High Speed with the LPC55S69.

 When tried creating the uVision project, I had an error:

```
 Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\fabio.silva\AppData\Local\Programs\Python\Python313\Scripts\progen.exe\__main__.py", line 7, in <module>
    sys.exit(main())
             ~~~~^^
  File "C:\Users\fabio.silva\AppData\Local\Programs\Python\Python313\Lib\site-packages\project_generator\__main__.py", line 67, in main
    return args.func(args)
           ~~~~~~~~~^^^^^^
  File "C:\Users\fabio.silva\AppData\Local\Programs\Python\Python313\Lib\site-packages\project_generator\commands\generate.py", line 70, in run
    build_failed, export_failed = r.get(timeout=20.0)
                                  ~~~~~^^^^^^^^^^^^^^
  File "C:\Users\fabio.silva\AppData\Local\Programs\Python\Python313\Lib\multiprocessing\pool.py", line 774, in get
    raise self._value
RuntimeError: Target cortex-m33 is not supported. Please add them to https://github.com/project-generator/project_generator_definitions
```

 I tried to create another projects using this processor, but, every time the same error!
 
 Someone, please, could help me to figure out this problem?

mathias-arm, 2024-12-05:
Try using `uvision5_armc6` as the toolchain.

fabio-ric-silva, 2024-12-06:
> Try using `uvision5_armc6` as the toolchain.

Mathis, first of all, thanks for your reply!
The problem is;

Using the progen generator, I always got that error!

I have installed the last version of uvision, updated all the libraries, and the problem persist!

I really don't now how to fix it!